FullControlXYZ/fullcontrol #1: Documentation for installation

WillAdams, 2023-03-21:
Please note minimum Python version:

>ERROR: Package 'fullcontrol' requires a different Python: 3.8.13 not in '>=3.9'

For Windows note that it will be necessary to install:

 - Python: https://www.python.org/downloads/windows/ --- https://www.python.org/ftp/python/3.10.10/python-3.10.10-amd64.exe
 - GIT: https://gitforwindows.org/ --- https://github.com/git-for-windows/git/releases/download/v2.40.0.windows.1/Git-2.40.0-64-bit.exe

and it would be very helpful to have a specific link to a complete, minimal working example, as well as notes on how the G-code will be previews.

AndyGlx, 2023-03-21:
Thanks for this! We'll update the readme once it's clear these instructions are working for other people. If anyone finds the above solution to work/not-work, please comment on this issue

AndyGlx, 2023-03-21:
By minimal working example, what would you want it to show that isn't in the quick demo (linked to in the main and docs' readmes) or the broad overview notebook, or the [example models](https://github.com/FullControlXYZ/fullcontrol/tree/master/models) (although I know these aren't 'minimal')? 
Or does it just need to be clearer that these models exist / how to use them?

WillAdams, 2023-03-23:
Pointing to the existing files would be fine, but I think a really short example which does something simple such as make a circle or hemisphere would be good --- it would give folks something to type in and read over and understand, and would communicate how math underlies the creation of objects with this tool.

fullcontrol-xyz, 2023-03-28:
Readme is now updated with advice to install the necessary software (anaconda/conda/python + git) and update to most recent versions if any issues are encountered. Also, emphasis is places on using the the colab notebooks initially to avoid the need for any installation. 

Readme also updated with a short extra introduction right at the top, with links to a minimal example and the online docs

closed 2023-03-28 by fullcontrol-xyz