FullControlXYZ/fullcontrol #3: [Q] Multiple Walls

Sineos, 2023-04-10:
I have described a profile via its point coordinates that represents the outer wall of a design. Is there an elegant way to turn these steps into multiple walls, each correctly offset in x and y by the extrusion width?

AndyGlx, 2023-04-10:
Hi, this is interesting functionality which can be very simple or a bit mathsy depending on exactly what you want. For the simplest geometry (a regular polygon) it can be done with one command already: a radial offset with move_polar(), which I can explain. 

Some other considerations:
- is the path a closed loop (starting and ending at the same point)?
- if the path is to the outside of the original one, should the corner angle to be maintained or would a fillet/chamfer be necessary (e.g. if you have very acute angles)?

Can you share your path (describe it or create a gist)? I'm not at a computer for a few days but I've already created functions for this kinda thing in unpublished designs, so I should be able to create a gist or 'lab' function quickly when I'm back at a computer

fullcontrol-xyz, 2023-04-10:
Hi, this is interesting functionality which can be very simple or a bit mathsy depending on exactly what you want. For the simplest geometry (a regular polygon) it can be done with one command already: a radial offset with move_polar(), which I can explain. 

Some other considerations:
- is the path a closed loop (starting and ending at the same point)?
- if the path is to the outside of the original one, should the corner angle to be maintained or would a fillet/chamfer be necessary (e.g. if you have very acute angles)?

Can you share your path (describe it or create a gist)? I'm not at a computer for a few days but I've already created functions for this kinda thing in unpublished designs, so I should be able to create a gist or 'lab' function quickly when I'm back at a computer

Sineos, 2023-04-10:
Thanks for your reply. Please see https://colab.research.google.com/drive/1TwfX--iDGd_hyYefp3tp8u_QLe_XPBTm?usp=sharing --> `# Create PA cube`
In the current use case (my first contact with FC; great work!) it is a pretty trivial closed loop polygon. I was just wondering if there is a more elegant way instead of repeating the entire steps again and manually applying the needed offsets. Something like:

`fc.move(steps, fc.Vector(x=extrusion_width, y=extrusion_width), copy=True, copy_quantity=walls)`

More generally speaking, I would guess that multiple walls are a pretty common thing for 3D printers but of course you are right, things can get complicated pretty quickly.

fullcontrol-xyz, 2023-04-10:
Ah that design is great! 

Because it's a regular polygon, you can do what you want easily with [fc.move_polar] (https://github.com/FullControlXYZ/fullcontrol/blob/master/docs/geometry_functions.ipynb) to create radially expanded/contracted paths 

For the move_polar function, set the centre point to be a fc.Point defined at the centre of the square, set rotation offset to be zero (no change) and set radial offset to be extrusion_width/sin(0.5*corner_angle_in_radians), and set it to copy, not just move. 

So, for you, radial offset is extrusion_width/sin(0.125*tau)

Consider offsetting by, say, 90% of extrusion width, to get a good lateral bond between extrudates

Let us know if it works

Sineos, 2023-04-11:
Thanks so much for the support. Alas, I'm hitting a problem here. Added:

```
# Generate walls
cube_steps_prepare_walls = fc.move_polar(cube_steps_prepare, fc.Point(x=mid_bed_x,y=mid_bed_y), 0, extrusion_width / math.sin(0.125 * math.tau), copy=True, copy_quantity=2)
```
and got:
```
[/usr/local/lib/python3.9/dist-packages/fullcontrol/geometry/polar.py](https://localhost:8080/#) in point_to_polar(point, centre)
     18 def point_to_polar(point: Point, centre: Point) -> PolarPoint:
     19     'find polar radius and angle (radians: 0 to 2pi) in XY plane of the given Point relative to the centre Point. return a PolarPoint, accessed with .radius and .angle)'
---> 20     r = ((point.x - centre.x) ** 2 + (point.y - centre.y) ** 2) ** 0.5
     21     angle = atan2((point.y - centre.y), (point.x - centre.x))
     22     return PolarPoint(radius=r, angle=angle % tau)

TypeError: unsupported operand type(s) for -: 'NoneType' and 'float'
```

Please see updated version under above link.

Sineos, 2023-04-11:
Indeed that did the trick for the TypeError. Thanks!
Still not quite where I would like to be:

[image]

The polar move causes a rotation instead of only a translation

fullcontrol-xyz, 2023-04-11:
I think it's doing rotation instead of radial translation. Probably just need to change the order of the parameters you're supplying to the move_polar function. Or name them explicitly. Play in a new notebook code cell with just a rectangle, a polar move function and plot transformation - you'll sort it quickly

Sineos, 2023-04-11:
I have to admit that this function confuses me. I had tried both ways and, i.e.

```
# Generate walls
cube_steps_prepare_walls = fc.move_polar(cube_steps_prepare, fc.Point(x=mid_bed_x,y=mid_bed_y), 0, extrusion_width / math.sin(0.125 * math.tau), copy=True, copy_quantity=2)
```

which is causing a rotation. And

```
# Generate walls
cube_steps_prepare_walls = fc.move_polar(cube_steps_prepare, fc.Point(x=mid_bed_x,y=mid_bed_y), extrusion_width / math.sin(0.125 * math.tau), 0, copy=True, copy_quantity=2)
```

which in fact does a translation BUT is causing a distortion:

### Exaggerated view 
Using:
fc.move_polar(cube_steps_prepare, fc.Point(x=mid_bed_x,y=mid_bed_y), **10, 0,** copy=True, copy_quantity=2)

[image]

### Detail view using Prusa Slicer a Gcode viewer
Using:
fc.move_polar(cube_steps_prepare, fc.Point(x=mid_bed_x,y=mid_bed_y), **extrusion_width / math.sin(0.125 * math.tau), 0**, copy=True, copy_quantity=2)

[image]

### Detail 1
[image]

The dark blue lines are nicely parall and on top of the lower layer

### Detail 2
[image]

The distortion is visible. The two "walls" are no longer parallel and no longer on top of the layer below. Same pillowing effect that can be seen in the "Exaggerated view"

### Detail 3
[image]

The starting points of the lines are being shifted, so no true parallel translation like I would expect for "walls"

fullcontrol-xyz, 2023-04-11:
Ahhh it's because you have points along the sides. The function is working fine but the maths assumptions for the method I described only works for a polygon with straight lines between the corner points (no extra points along the straight lines). In a few days, when I'm back at a computer, I'll look into the other option I described in my first post, that I've previously implemented for paths that aren't regular polygons

fullcontrol-xyz, 2023-04-11:
Haha you don't need a maths degree(!). My engineering degree would be scoffed at by mathematicians. All the maths in FullControl is actually relatively low-level in terms of technical maths, but just implemented for non-trivial things. But being comfortable with maths is really advantageous and opens up lots of opportunities. Some things are kinda "you like it or you don't" but I hope other bits of FullControl will make it accessible to non-mathsy people, like if I add an 'offset_path' function that does the maths for this thread behind the scenes. In terms of who to target, it'll be a bit of an evolving process. We'll see what sorta people are using it for what sorta thing!

fullcontrol-xyz, 2023-04-18:
I've added a function called offset_path to FullControl

It's currently located in the FullControl lab. and demonstrated in the [lab tutorial notebook](https://colab.research.google.com/github/FullControlXYZ/fullcontrol/blob/master/docs/colab/lab_geometry_colab.ipynb)

[image]

[image]

closed 2023-04-18 by fullcontrol-xyz