FullControlXYZ/fullcontrol #4: [FR] Colors transition off

Sineos, 2023-04-12:
I'm using `color_type='manual'` to highlight property changes (e.g. speed) in the design. The transition between the different colors make the result look wrong. 
Is there a way to have a hard cut over to a new color?

fullcontrol-xyz, 2023-04-12:
It's because the lines have a gradient between the point's colours. Try adding two points with different colours either directly overlapping or close to each other so this gradient appears to be instant

Sineos, 2023-04-12:
Hmm, OK, this works but I do not think that changing colors warrants cluttering the design with additional points. Unfortunate but not a deal breaker. Thanks.

fullcontrol-xyz, 2023-04-14:
I made a minor correction to the code. You can now add a point with only the color attribute defined (no values of x y z entered). This suits the overall approach of FullControl to define the sequence of how things change.... 

1. if you define a color change **before** an xyz movement, the color change happens before the line
1. if you define a color change **after** an xyz movement, the color change happens after the line
1. if you define a color change **at the same time** as an xyz movement, the color change happens during the line

the relevant [tutorial notebook](https://colab.research.google.com/github/FullControlXYZ/fullcontrol/blob/master/docs/colab/state_objects_colab.ipynb) has been updated to demonstrate this

closed 2023-04-14 by fullcontrol-xyz

Sineos, 2023-04-15:
[image]

Works nicely. Thanks a lot!