FullControlXYZ/fullcontrol #10: Documentaion / Reference website

dicksondickson, 2023-04-28:
Thank you for all the work put into creating the examples in the python notebooks. 

I'd imagine you'd be adding much more functions to FC. Going forward, may I suggest creating a wiki?
I always found the doc pages for Processing  [](https://processing.org/reference/ )to be very convenient to use.

Thank you!

fullcontrol-xyz, 2023-04-29:
Thanks that looks really clean and nice. And will be necessary as more fun functions get added. We're also considering automatic generation of the tutorial notebooks. Hopefully it'll be a nice simple framework... when a new function gets added, it's also accompanied with a standalone bit of documentation which updates everything else as appropriate. This will likely be done after python source code is simplified a bit (planned).

fullcontrol-xyz, 2023-08-04:
added to roadmap - closing issue

closed 2023-08-04 by fullcontrol-xyz