FullControlXYZ/fullcontrol #11: SVG import

dicksondickson, 2023-04-28:
SVG import function will be awesome. I want to integrate typography and other shapes to use in FC to create some cool designs.

I think this will be very useful for those using FC with a laser cutter, pen plotter, and vinyl cutter(?).

Thank you!

fullcontrol-xyz, 2023-04-28:
I've made a [gist](https://gist.github.com/fullcontrol-xyz/fe80b7ae8af1f80eb601c3c1f480d8fe) to do this

You just need to add a few lines to open an svg file instead of writing the svg string directly in python code (and to upload the file to google colab if using it)

[image]

Let me know if this does what you need or if there are any problems. If it seems like the function works well after a couple of people have commented, I'll add it to FullControl lab

dicksondickson, 2023-04-29:
This is awesome thank you!
As a designer this function opens up more options for me.

The code works, but the paths are flipped like this:

[image]

ChatGPT helped me modify the code and now the results are as expected:

[image]

**Here is the modified code:**

`def get_point_at(path, distance, scale, offset, height):
    pos = path.point(distance)
    pos += offset
    pos *= scale
    return pos.real, height - pos.imag  # invert the y-coordinate

def points_from_path(path, density, scale, offset, height):
    step = int(path.length() * density)
    last_step = step - 1

    if last_step == 0:
        yield get_point_at(path, 0, scale, offset, height)
        return

    for distance in range(step):
        yield get_point_at(path, distance / last_step, scale, offset, height)

def points_from_doc(doc, density=5, scale=1, offset=0):
    height = float(doc.getElementsByTagName("svg")[0].getAttribute("viewBox").split()[3])
    offset = offset[0] + offset[1] * 1j
    points = []
    for element in doc.getElementsByTagName("path"):
        for path in parse_path(element.getAttribute("d")):
            points.extend(points_from_path(path, density, scale, offset, height))

    return points

def svg_to_points(svg_str: str, z_height: float) -> list:
    doc = minidom.parseString(svg_str)
    points = points_from_doc(doc, density=1, scale=5, offset=(0, 5))
    doc.unlink()
    return [fc.Point(x=pt[0], y=pt[1], z=z_height) for pt in points]
`

**And for reading svg file in the notebook:**

`svgFile = "type.svg"

with open(svgFile, 'rb') as f:
    svg = f.read().decode('utf-8')`

fullcontrol-xyz, 2023-04-29:
Ah brilliant!! 
I'll update the original gist when I can. I can see there are lines connecting the letters in the printpath preview in your post. Presumably those lines don't appear in the SVG itself? I'll try to find a way to set these as travel. Should be simple. Can you share your svg as an attachment to reply to this? Or as a gist or something?

dicksondickson, 2023-04-29:
> Presumably those lines don't appear in the SVG itself? I'll try to find a way to set these as travel.

Yes you are correct, each letter should be standalone and not connect. Thank you so much!

Here is the SVG file:

[image]

I think this will SVG import example will serve as a good reference for newcomers to FC.

fullcontrol-xyz, 2023-04-30:
Thank you. In terms of size, is that svg somewhat 'normal'? Did you choose size or anything? I can see it's big on the print bed (as was my example) but want to make the default values in the function reasonable for desktop printing at maybe 1/10th the size of our examples

dicksondickson, 2023-04-30:
Now that you mentioned it, the size is definitely not correct. The width of the text should be 116mm.
I created the SVG in Adobe Illustrator and exported as SVG. Do you think Illustrator is exporting the SVG at an incorrect scale?

dicksondickson, 2023-04-30:
In Illustrator I set my units to mm. When exporting to SVG, it converts to some unknown scale.

I changed my units to pixels and now the exported SVG viewbox corresponds to the width of the text (116px) but in the plot its showing the same size.

Here is the new updated SVG:

[image]

dicksondickson, 2023-04-30:
Ok its working as expected now. Here is the updated code:

``
from svg.path import parse_path
from xml.dom import minidom

# from https://stackoverflow.com/questions/69313876/how-to-get-points-of-the-svg-paths

def get_point_at(path, distance, scale, offset, height):
    pos = path.point(distance)
    pos += offset
    pos *= scale
    return pos.real, height - pos.imag  # invert the y-coordinate

def points_from_path(path, density, scale, offset, height):
    step = int(path.length() * density)
    last_step = step - 1

    if last_step == 0:
        yield get_point_at(path, 0, scale, offset, height)
        return

    for distance in range(step):
        yield get_point_at(path, distance / last_step, scale, offset, height)

def points_from_doc(doc, density=2, scale=1, offset=0):
    height = float(doc.getElementsByTagName("svg")[0].getAttribute("viewBox").split()[3])
    offset = offset[0] + offset[1] * 1j
    points = []
    for element in doc.getElementsByTagName("path"):
        for path in parse_path(element.getAttribute("d")):
            points.extend(points_from_path(path, density, scale, offset, height))

    return points

def svg_to_points(svg_str: str, z_height: float) -> list:
    doc = minidom.parseString(svg_str)
    points = points_from_doc(doc, density=2, scale=1, offset=(0, 0))
    doc.unlink()
    return [fc.Point(x=pt[0], y=pt[1], z=z_height) for pt in points]
``

Scale var was set differently in 2 places.

So the workflow is, in whatever program you are creating the SVG, work in pixel units, export/save as SVG then load into FC.

fullcontrol-xyz, 2023-04-30:
Great stuff! I expect people will use lots of different svg-generation software, with a range of potential size-conversion strategies, so I'll make it easy to adjust scale when calling the function in FullControl. I'll also do some tweaks to make sure the code can be imported nicely when using FullControl in jupyter notebooks and in plain python scripts

dicksondickson, 2023-04-30:
Awesome thank you!

ES-Alexander, 2023-05-01:
@dicksondickson neat idea - I'm sure this will be useful to lots of people! :-)

Just an FYI, you can get GitHub to format your code nicely by specifying the language at the start of the code-block, e.g.

````
```python
def function(var):
    pass # do something
```
````
becomes
```python
def function(var):
    pass # do something
```

fullcontrol-xyz, 2023-08-04:
added to roadmap - closing