FullControlXYZ/fullcontrol #8: Safety Check Function?

dicksondickson, 2023-04-28:
[image]

The above model was created with FC Web using the ripple texture demo and printed on a Prusa MK3S+. 
The flat surface that you see is caused by the model exceeding the printable area. 

Would it possible to integrate a "safety check" function where the design is evaluated against the printer profile such as the bedsize and other hardware limits (temperature, travel speed)?

Thank you for making FC!

dicksondickson, 2023-04-28:
Also helpful for non-planar printing so I don't accidentally crash my extruder into the model. ;)

fullcontrol-xyz, 2023-04-29:
Yeah good idea. This will be easy once we (hopefully) link up with existing printer libraries (e.g. in open-source slicers)

fullcontrol-xyz, 2023-08-04:
added to roadmap, linked to [issue 21](https://github.com/FullControlXYZ/fullcontrol/issues/21)

closed 2023-08-04 by fullcontrol-xyz