FullControlXYZ/fullcontrol #14: GCODE save filename date and time

dicksondickson, 2023-05-04:
Is there is a way to turn off the automatic appending of the date and time to the gcode file?

I would like to have FC export the gcode the same as the design name only without the date and time so it fits within my workflow.

Thank you!

fullcontrol-xyz, 2023-05-04:
``` python
    if gcode_controls.save_as != None:
        filename = gcode_controls.save_as + datetime.now().strftime("__%d-%m-%Y__%H-%M-%S.gcode")
        open(filename, 'w').write(gc)```

fullcontrol-xyz, 2023-05-04:
You can save it as any filename you like with an extra line of code in your design... see near the top of the [gcode_controls notebook](https://githubtocolab.com/FullControlXYZ/fullcontrol/blob/master/docs/colab/gcode_controls_colab.ipynb)

I think this is the best option... you could also add a couple of lines to save a file of the same name with your design settings in to have a really nice record of what you've done. And then go to the next level and optionally import your settings from that file rather than manually defining them :D

Alternatively, if you clone the fullcontrol repo, you can edit the module fullcontrol/gcode/steps2gcode.py 

relevant code:
```python
      if gcode_controls.save_as != None:
          filename = gcode_controls.save_as + datetime.now().strftime("__%d-%m-%Y__%H-%M-%S.gcode")
          open(filename, 'w').write(gc)
```

closed 2023-05-04 by fullcontrol-xyz

dicksondickson, 2023-05-04:
Thanks for the tip! Exactly what I was looking for. 

Just from a usability standpoint, I do think "datetime.now().strftime("__%d-%m-%Y__%H-%M-%S.gcode")" should be optional and added by the user in their notebook / python script.

While I've been using FC for the past few days, I have a over a hundred .gcode files because I forgot to comment out the save to gcode block in my notebook while working on a design.

Thank you again

fullcontrol-xyz, 2023-05-05:
Great, and thanks for your suggestion. I've created a reddit poll to see what most people prefer as the default option

[reddit poll](https://www.reddit.com/r/FullControl/comments/138bwm7/poll_preference_about_default_filename)