FullControlXYZ/fullcontrol #16: SVG Export

dicksondickson, 2023-05-10:
I've been getting some really cool shapes in FC and want to export it to SVG.
Since FC can do SVG import now, is it possible to export the paths/points in FC to SVG?

Thank you!

aapolipponen, 2023-05-10:
Matplotlib pyplot has that. Technically you could convert the fc.Points for matplot and export them trough that but svg export could be cool. Also svglib could be used.

fullcontrol-xyz, 2023-05-10:
Depending how perfect you need the svg format to be, and the complexity of your path, you might be able to just create a string and save it as a .svg file. This is easy if you have a single continuous path.

E.g. the following is a simple SVG file for a triangle
```
<svg xmlns="http://www.w3.org/2000/svg">
<path d="M100,100 L300,100 L200,300 L100,100" style="stroke:blue;stroke-width:2px; fill:none" />
</svg>
```
All you'd need to do is inspect the fullcontrol steps and extract X and Y values from the points and formal a string like 
```
f'M{x1},{y1} L{x2},{y2} L{x3},{y3} L...'
```
if your list of steps has some travel moves without extrusion, you'd want to use M instead of L in the string to 'move' to the next point rather than plotting a 'line' to it. Since fullcontrol already does calculations for that sort of thing when generating the built-in plot, it might be easiest to get the raw plot data from fullcontrol using the raw_data=True option in PlotControls and then inspect that in order to create the svg string.

dicksondickson, 2023-05-11:
Thanks for the pointer on using the raw plot data! Going to look into that.
I discovered some cool patterns while experimenting with FC and want to export that out and use it as graphic design elements thus the reason for wanting to export.