FullControlXYZ/fullcontrol #24: [infrastructure] Add issue and pull request templates

ES-Alexander, 2023-06-09:
Github has quite extensive support for [Issue and PR templates](https://docs.github.com/en/communities/using-templates-to-encourage-useful-issues-and-pull-requests/configuring-issue-templates-for-your-repository), which can be quite helpful for recommending a particular structure for suggestions and bug reports/fixes and the like, while also automatically adding some labels to them.

This isn't particularly critical for a small project, but guided steps lower the barrier for entry to newcomers making valuable suggestions and contributions, and becomes more important as the number of submissions grows.

If it's of interest, [these](https://github.com/bluerobotics/BlueOS-docker/tree/master/.github/ISSUE_TEMPLATE) are the issue templates my work uses.

AndyGlx, 2023-06-10:
Thanks, these will be very useful. Especially for minor bug report or if people experience unexpected behaviour in FullControl

ES-Alexander, 2024-02-14:
Noting that issue templates have been added in #51, but I'm leaving this open for now since
1. there aren't yet any pull request templates
2. issue templates were requested separately in #50, which _has_ been resolved