FullControlXYZ/fullcontrol #30: Fun with projective perspective (+ seeds of a new logo/icon?)

ES-Alexander, 2023-07-05:
I've seen some cool projective perspective examples recently where depending on the angle an object is looked at it can show different things. I had some offline time on a flight and decided to try my hand at it for "Full Control", and thought the result was worth sharing:
https://gist.github.com/ES-Alexander/84a0f12d097d409bf70fed07f252ae91

Currently this is a relatively low level example of tube mesh generation, but if we wanted to we could turn it into a gcode example that then goes through the standard plotting process, with the caveat that although it looks cool it's likely not particularly practical as a 3D print.

---

No idea if there's interest in changing the project logo, but my idea there is it could be fun/interesting to have an icon/image that starts with `F_U_L_L` at the top, then shows one or more transition angles below, followed by `CONTROL` at the bottom. In my head that (for some reason I don't understand) feels like a nice indicator of what kinds of things can be created/done with the project.

e.g. something like the following (left ones have a transparent background, bottom ones have thicker letters and shrunken transition displays):

[image]&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;[image]

<br>
[image]&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;[image]

AndyGlx, 2023-07-05:
Ah this is so cool. I couldn't initially tell what the images were showing, but I think an animation would be really clear (continually running back and forth like a gif / Instagram-boomerang). The logo could always be a gif. The eagle logo has kinda been dropped at the moment. I created some ideas but generally didn't like them as much as the eagle. So it might come back. But this text logo would be good alongside or interchangeable with the eagle.

P.s. Still evaluating STL output. Just slow cos working on several FullControl things in parallel.

ES-Alexander, 2023-07-05:
> Ah this is so cool. I couldn't initially tell what the images were showing, but I think an animation would be really clear (continually running back and forth like a gif / Instagram-boomerang). The logo could always be a gif.

Fair enough. Sounds fun, although I'm not sure how challenging it would be to generate an animation/gif automatically. Doing it by hand (e.g. from screenshots) is definitely possible, but would be somewhat time consuming. The screen recording in the gist was pretty straightforward to make, but it'd be nice to have something smoother (and without the plotly background) if we were going to actually use it for something.

Not sure what would be better out of bouncing back and forward vs a continuous rotation that slows at the two upright word positions (and zooms more quickly through the upside down positions). If we set up something automatic that would likely be pretty easy to test, potentially with some custom tweening functions.

> P.s. Still evaluating STL output. Just slow cos working on several FullControl things in parallel.

No worries - I'm still planning to work more on it to check off more of the checklist, just not sure when I'll have time to do so at this stage. If I can't find time during my trip then I might need to come back to it in a few weeks when I'm back home.

ES-Alexander, 2024-02-28:
[image]

ES-Alexander, 2024-03-03:
@fullcontrol-xyz not sure if you've seen this^ yet.
Should I preferentially tag this account instead of the AndyGlx one? :-)

fullcontrol-xyz, 2024-03-03:
Sorry, I actually have a few chats I'm going to respond to (left as unread emails). Just been slow, but on Friday I cleared my work-email backlog for the first time since autumn!! So I'm happy I'll get some more time for this stuff now. With this logo it does look v cool. Are you able to create a variant with constant line width, that's a bit chunkier, and only black? And sharp joints/ends like the 'T'. I can see colour bring included, but also wanna see the simple version that would go on my mostly black-and-white videos.

In terms of paused, I think it's not actually a bad thing if people need to look a couple of times to understand the words. Quite a few social media posts do that deliberately to attract attention and stop the user scrolling past

ES-Alexander, 2024-03-03:
> Sorry, I actually have a few chats I'm going to respond to (left as unread emails). Just been slow

Yeah no worries - didn't mean to rush you, just wasn't sure if you'd been notified of this one since it's further down and I tagged a different account :-)

> ... on Friday I cleared my work-email backlog for the first time since autumn!!

Wow, nice work! Backlogs tend to predominantly grow in my experience... :-P

> Are you able to create a variant with constant line width, that's a bit chunkier, and only black? And sharp joints/ends like the 'T'.

Hopefully this is something like what you were after?
[image]
[image]

I also did grey 666, since that's about as dark as you can go while having the shading still properly visible:
[image]
[image]

I do find that the varied width segments on the curves help to give a better sense of the 3D-ness of the structure than the constant width options, while also show-casing that FullControl allows width control (not just path control), but perhaps neither of those are essential features for the use-cases you have in mind.

Let me know if there's anything else you want generated, or if you want you can try generating some ideas yourself using the code in the gist (if you do `fig.show()` instead of the image writing loop at the bottom then you can look at the model, which is likely the quickest way to iterate ideas) :-)

AndyGlx, 2024-03-18:
Thanks for these. They are looking much more aligned with my normal styling. I can also see the value of the other ones, but do try to be as simple/minimal as absolute possible. But these are really cool. I got your gist to run, but it's currently set up for the old coloured and variable-width design. Do you have the settings you used for the above designs? Also, if you could chunkify it, that'd be great - or point me to where I could control the tube dia.... maybe making the tube diameters 20-50% wider. I know that will mess up spacings and stuff, so perhaps don't worry about spacings. Can get the preferred letter proportions and then do spacing after.

Thanks,

Andy