FullControlXYZ/fullcontrol #31: Pydantic version issue

basti00, 2023-07-05:
**Issue Description:**
The project is not working with pydantic version 2.0.2. I guess v2 and above. At import (`import fullcontrol as fc`) the following ValidationError is thrown:

```log
  File "~\Projects\fullcontrol\fullcontrol\gcode\state.py", line 24, in State
    point: Optional[Point] = Point()
  File "~\Projects\fullcontrol\.venv\lib\site-packages\pydantic\main.py", line 150, in __init__ 
__pydantic_self__.__pydantic_validator__.validate_python(data, self_instance=__pydantic_self__)
pydantic_core._pydantic_core.ValidationError: 3 validation errors for Point
x
  Field required [type=missing, input_value={}, input_type=dict]
    For further information visit https://errors.pydantic.dev/2.1.2/v/missing
y
  Field required [type=missing, input_value={}, input_type=dict]
    For further information visit https://errors.pydantic.dev/2.1.2/v/missing
z
  Field required [type=missing, input_value={}, input_type=dict]
    For further information visit https://errors.pydantic.dev/2.1.2/v/missing
```

**Current Environment:**
Commit 0e89b2f10805426e788379b1c93ad3c677a012d6
Pydantic version: 2.0.2
Other dependencies: plotly, numpy

**Expected Behavior:**
No error at import.

**Steps to Reproduce:**
Install Pydantic version 2.0.2.
Run `import fullcontrol as fc`
Should cause a ValidationError.

**Fix Proposal:**
Include the recommended version ranges in the `requirements.txt` file (eg. pydantic<2.0.0) or update the type definitions so that pydantic v2 is happy. I tested and verified it is working with the versions pydantic-1.10.11 and pydantic-1.10.5.

If you require any further details, please let me know.

btw. really love your work, can't wait to try it, thank you!

lokithus, 2023-07-17:
@basti00

THANK YOU SO MUCH FOR THIS. I was having problems to make it work and seeing your post helped me a lot.

Also, thanks to all the programmers working with FullControl, this program will really revolutionize 3D printing.

fullcontrol-xyz, 2023-07-18:
I just realised I didn't reply to this @basti00. Thanks for raising it and explaining the steps. Will definitely fix it ASAP with the temporary solution in requirements.txt (and pyproject.toml). Then a better solution later 

And thanks for all the positive feedback ☺

fullcontrol-xyz, 2023-08-04:
implemented!

fullcontrol-xyz, 2023-08-04:
I've reverted one aspect of the merge due to an issue with using model_validator instead of root_validator in base.py. The new code that you added is included in base.py but commented out.

Your updated code didn't work for older versions of pydantic (including google colab, which has pydantic v1.10.12). So I've added the pydantic requirements to be <2.0 in requirements.txt and pyproject.toml and gone back to the old code in base.py. All the other changes you did are kept as they were, so this should be a minor code revision to get it working.

Did you successfully get fullcontrol to work in older versions of pydantic as well as newer versions?

fullcontrol-xyz, 2024-08-01:
This is now implemented in commit [bf8066836df2c0adc2e7d82ae9dc794dbb9dfce0](https://github.com/FullControlXYZ/fullcontrol/commit/bf8066836df2c0adc2e7d82ae9dc794dbb9dfce0)

Thanks for raising the issue!
requirements.txt and pyproject.toml have been updated to allow all versions of pydantic. The updated fullcontrol code checks which version of pydantic is installed and implements the appropriate function for the installed version.

closed 2024-08-01 by fullcontrol-xyz