FullControlXYZ/fullcontrol #34: [feature] Rounded Square wave

Keelan-Moore, 2023-07-18:
I made a new wave type, its basically a square wave with circular peaks and troughs. It was just for a small test project I was trying but figured it may have use elsewhere so I have made a [gist for the wave](https://gist.github.com/Keelan-Moore/62836d4792f0ed9d0ac84e064d266176) and included example output below showing that it has the same flexibility as the existing square wave. 

[image]

Let me know what you all think

lorenuars19, 2023-07-25:
Could even be used as infill, that would be a very nice addition

Keelan-Moore, 2023-07-27:
Yeah! Cool idea!  It could be the basis for a gyroid style infill by changing the phase with height

AndyGlx, 2023-08-04:
Yeah that's really useful thanks! I know people who print microfluidics devices and this will definitely be interesting to them. They print the lines in ABS, then encapsulate in a gel, then dissolve away the ABS to leave perfect smooth tiny channels. Fluid flows through those channel and magic microfluidic stuff happens. The rounded corners will obviously be nice for fluid flow. I'll have a think about how to add this into the existing waves. It might be easiest for now to just add it to the list and include it in the exiting demo documentation. But at some point that will need to change when there are too many different types of waves

closed 2023-08-08 by fullcontrol-xyz