FullControlXYZ/fullcontrol #37: Printing between fc.move repeats

B-Carver-Creatives, 2023-08-08:
When using fc.move to repeat a part (or segment of a part in this case) in the x,y plane, a print line is created between the parts. 

Is the a way to make this a travel move? 

I have the list of commands being repeated ending with
rep.append(fc.Extruder(on=False))

it ends up looking something like this 
[image]

Many thanks

fullcontrol-xyz, 2023-08-08:
I would do something like this:

``` python
steps = []
steps1 = [...some steps...]
steps2 = fc.move(steps1, ...)
steps.extend(steps1)
steps.extend(fc.travel_to(steps2[0]) # travel to first point in the next list of steps
steps.extend(steps2)
```

You can do that code much more concisely if you want by using `steps1 + fc.travel_to(steps2[0]) + steps2` or similar

B-Carver-Creatives, 2023-08-09:
Conceptually that make scene.  However, when I tried it ( using the .extend method and list addition), the extruder is turned off for the travel move, but does not turn back on for the subsequent print steps. 
[image]

I tried adding fc.Extruder(on=true) at the beginning of step list and traveling to the point in line [1]. This turned the extruder on for the copied print but also for the travel move

fullcontrol-xyz, 2023-08-09:
If you use the travel_to function, it'll turn off and and turn on extrusion for you. You don't need to manually add any Extruder(on=?) objects to your list of steps. Check out the travel_to example in the other functions tutorial. Also make sure your 'move' command is not copying the geometry. 

Or if still struggling, create your list of steps sequentially. First add the first circle section, then turn extrusion off, the move to the first point of the second circle section, then turn extrusion on, then add all the steps for the second circle section. Do similar when going to the first circle section of the second layer.

B-Carver-Creatives, 2023-08-09:
Ok I got It figured out. 

I was copying during the move ( I wasn't thinking about how I was copying by defining it as a new list) 

I think this was causing a bunch of weird results in the plot. Not exactly sure what was causing the issues I was seeing, 

Regardless, organizing the travel moves as a separate list and then adding the lists together fixed the problem. 

Many thanks for your help!

fullcontrol-xyz, 2023-08-09:
Great! And remember your travel currently doesn't include any retraction. You can use `fc.PrinterCommand(id='retract')` and `fc.PrinterCommand(id='unretract')` before and after the travel move if desired. Or explicitly define the retraction (negative volume) or unretraction with `fc.StationaryExtrusion(volume=5, speed=1000)`

closed 2023-08-09 by fullcontrol-xyz