FullControlXYZ/fullcontrol #38: Z Travel Speed

B-Carver-Creatives, 2023-08-11:
Is there a way to set the travel speed to a different value for the z axis? 

Ive got a unique printer and it can run at pretty high travel speeds (18000 mm/min) but the z axis needs to move considerably slower (1000 mm/min) 

Thanks 
-Ben

fullcontrol-xyz, 2023-08-11:
There's no built-in method for that. I think you'd need to set travel speed to be 1000 immediately before the Z movement and then revert it immediately afterwards. There are various other ways to achieve the same thing but that's a pretty simple way.

B-Carver-Creatives, 2023-08-11:
Gotcha, thanks 

work arounds are fine but I wanted to check if there was something built in.
Thanks!

fullcontrol-xyz, 2023-08-11:
Objection (!), I wouldn't call it a work around 🙃 it's just the principle of FullControl, that you describe any changes to *state*. There could be a function like fc.travel_z(), but it would likely be a shortcut to implement the thing I've described above. 

However, if this is a super common requirement, it could be the case that FullControl had a travel_speed_z and travel_speed_xy instead of just travel_speed, but I expect there are just too many other situations when speed might be changed. And it wouldn't be feasible to include them all.

If you want to automate it, you could loop through all your steps after creation, find any that are fc.Points with only Z defined, then replace them with three steps: change speed, point, change speed.

B-Carver-Creatives, 2023-08-11:
Ok I see what you mean about breaking it all down into charges to the state. 

I was envisioning something like the travel_speed_Z and travel_speed_XY, though I understand if it would create problems elsewhere. Just checking. 

The reason I was looking for it is that the Z axis mechanics of printers often have a different max speed than X and Y motion. So if you finish a print and have your end gcode go to a parking position (ex fc.Point(x=0 y=0 z=200), it would be nice to have the Z axis got at a slower speed. 

 Would it be possible to add a manual gcode max speed in Z (M203 Z1000) command to my starting gcode? or would this confuse other things in full control?

Thanks

fullcontrol-xyz, 2023-08-12:
Yeh I totally understand the value and wanna make everything as seamless as possible. It's great that you're asking so we can keep trying to make FullControl as relevant to everyone as possible. 

Adding the max Z speed GCode to the start is actually the simplest way. You could also put it at the start of the fullcontrol design (ManualGcode object in the list of steps). That actually is a genuine workaround cos you're telling your printer to ignore/override the instructions for state-changes in the fullcontrol design. But I like it!

B-Carver-Creatives, 2023-08-12:
Ok cool, Ill give that a try. 

Many thanks for answering my multitude of questions this week! Ive been looking forward to the FC python release for a while, but am only now able to devote the time to learning it.

closed 2023-10-11 by fullcontrol-xyz