FullControlXYZ/fullcontrol #44: Repo for the fullcontrol.xyz website?

kumavis, 2023-10-13:
Is there a repo for the fullcontrol.xyz website? I'd like to make some contributions

fullcontrol-xyz, 2023-10-14:
Ah exciting to hear you can contribute! There isn't a public repo right now but it is planned to be public. Dirk and Andy will discuss options early next week... e.g. making it public or inviting private contribution (if quicker/easier in the short term). 

FYI, I've made some minor changes recently but having made them live yet. So you might see some differences in the website soon, or in the repo.

Shall reply again soon

fullcontrol-xyz, 2023-12-12:
A quick update on this. I have tidied up the code a little, but have been swamped since Oct with uni teaching and not been able to do everything I wanted on FullControl. This code will still be released in its own repo, but I'm having some time off work/coding for a while over xmas, so it will be next year before that happens