FullControlXYZ/fullcontrol #46: Addressing Offset Path Function Behavior Near Corners

sweikert, 2023-11-09:
The current implementation of the offset path function exhibits undesired behavior near corners, where points that are close together result in an offset that may not align with user expectations. This is primarily due to the function offsetting each point normal to the instantaneous path direction.

[image]

fullcontrol-xyz, 2023-11-09:
I love this preview! Using the svg from this gist to handle an svg exported from illustrator? Or a revised version of that gist? 

https://gist.github.com/fullcontrol-xyz/fe80b7ae8af1f80eb601c3c1f480d8fe

Yep this happens when points are close together near a corner. It means the offset path function's code is nice and simple. One solution is to go through the  offset path after creating it, segment by segment, and check for intersection with any segments in the near future of the path. E.g. Look 10 segments ahead. There are already functions to check for intersection between lines in the FullControl lab. Once you found out which segments intersect, you can replace the points in between with the intersection point. It'll take a little extra thought at the very beginning/end of the path, maybe looping through the end of the list back to the start somehow.

I'll look into the code sometime, but it might not be for a while. Please share if you develop your own. I can add it to lab.

sweikert, 2023-11-09:
I have a customized version of the Gist. I want to be able to read individual objects from the SVG. Then I can specifically control how the path is printed later. 

And my SVGs come from Illustrator. This is the easiest way for me to work with the points for paths.