FullControlXYZ/fullcontrol #49: [FEATURE REQUEST] Add security policy

darigovresearch, 2023-12-10:
**Is your feature request related to a problem? Please describe.**
There appears to be no way to understand what to do if there is a security bug

**Describe the solution you'd like**
Add a policy to the security tab

**Describe alternatives you've considered**
Can be in the readme but the security tab is probably best

**Additional context**
NA

fullcontrol-xyz, 2023-12-12:
Thanks you. I've added a simple security policy. Please let me know if you think I've overlooked anything.

darigovresearch, 2023-12-16:
Looks good, you might not want to use your exact email to avoid spam email parsers picking up your email

That's why you usually see something like info [at] fullcontrol.xyz

fullcontrol-xyz, 2023-12-17:
I actually share that email address widely on social media to try to encourage engagement with one click, so I doubt it will make much difference to include it here, but thanks for the suggestion 👍

closed 2024-02-26 by fullcontrol-xyz

darigovresearch, 2024-03-02:
You're very welcome and thanks again for adding the security policy!