FullControlXYZ/fullcontrol #58: [BUG] PlotControls style='tube get errors line not

Birhan-t, 2024-01-06:
this line;
fc.transform(steps, 'plot',fc.PlotControls(style='line')) works perfectly but,
fc.transform(steps, 'plot',fc.PlotControls(style='tube')) couse errors;

TypeError                                 Traceback (most recent call last)

[<ipython-input-7-903bc4bfd137>](https://localhost:8080/#) in <cell line: 34>()
     32 
---> 33 fc.transform(steps, 'plot',fc.PlotControls(style='tube'))
     34 

[/usr/local/lib/python3.10/dist-packages/numpy/lib/function_base.py](https://localhost:8080/#) in diff(a, n, axis, prepend, append)
   1445     op = not_equal if a.dtype == np.bool_ else subtract
   1446     for _ in range(n):
-> 1447         a = op(a[slice1], a[slice2])
   1448 
   1449     return a

TypeError: unsupported operand type(s) for -: 'float' and 'NoneType'

code link: https://colab.research.google.com/drive/1fwnmY33js1wj_RPKJqOIKz8EYxWbbSyh?usp=sharing

Birhan-t, 2024-01-07:
I think i found problem if add travel without any axis coordinates like steps.extend(fc.travel_to(fc.Point(z=5))) couse error.

fullcontrol-xyz, 2024-01-20:
I can't access the colab code (needs to be made public) but I'm interested to see what caused this cos I'm planning to add to automatic error checking in the future and I don't think your solution is generally true. For example, I just added the following two lines to the design_template model at it worked fine for line and tube plots... preview showed the additional travel line and printed line:

``` python
steps.extend(fc.travel_to(fc.Point(z=50)))
steps.append(fc.Point(z=60))
```

Birhan-t, 2024-01-23:
Hi, sorry for wrong sharing, try
https://colab.research.google.com/drive/1SkYellhPvEV_uXE19GmuwD8Ysik1ZgNm?usp=sharing

i commented code like this #=> for generate error comment this line
thanks.

fullcontrol-xyz, 2024-01-24:
Aha, the issue not really to do with the travel_to function, but rather due to the first point in your list of steps not having all of x y and z defined. Because you're using the travel_to function at the beginning of your list of steps, the first point in your list of steps ends up being the point you provide to the travel_to function. 

FYI, you don't need to use travel_to at the start of your list, you can choose the 'travel' primer in GcodeControls initialization_data.

I know this design-issue/requirement could be communicated to the designer - e.g. an error message saying the first point must have all of xyz defined - and do intend to add checks like that soon.

Thanks for following up

Birhan-t, 2024-01-26:
Thank you for the explanation. However, the actual code is quite long and I thought I found the problem, so I corrected the error places myself. The example I gave is just a short piece that repeats the error. In the original code, in the first line of steps, travelto x y and z are included. Since I am not very familiar with Python, I took the basic code from the examples on github and when I saw that they only used the needed  x y z parameters, I wanted to use it that way. Anyway, travelto was only needed when the layer was complete, as the code developed, I needed to make jumps without changing the layer, it was easier to write without z. In the end I don't think this is the problem. The error is not the error message of the fullcontrol function but of python, so it was quite difficult for me to understand the problem. it would be nice if the fc.transform function replaced it with a default value or gave a warning when a required parameter was missing :)

I had a similar situation when I added a manual g-code but I couldn't reproduce it. 
Thank you for your interest,
best regards

fullcontrol-xyz, 2024-03-17:
Yep I am definitely going to add some built-in checks to the transform function. There are some problems I always introduce accidentally even after using fc for a couple of years. Including:
- adding a list of points to steps by append instead of extend
- not fully defining the first point
- using partially defined points with move_polar(), which can only calculate rotations if by X and Y values are known
- and a few more