FullControlXYZ/fullcontrol #60: [QUESTION] How can I create infill within a custom geometry?

Timothee-Leblond, 2024-01-17:
**What are you trying to do?**
I'd like to create a pyramid (3 faces) with an infill, making possible to adjust the dimension of this pyramid without breaking the infill. All of this in a continuous path. 

**What have you done so far?**
I succeeded creating the first two layers of the pyramid but struggle with the next layers as followed:

[image]
[image]

**What do you still need to do?**
I'd to complete the pyramid and getting that continuous path.

**What is your question**
I'm looking for technical advice using FullControl, such as which feature is better to use in that case and how to achieve it.

Thank you!

Tim

fullcontrol-xyz, 2024-01-20:
Hi Tim,

Would you consider switching to the python version of FullControl? 
Check the README on the home page of this github repo for more info about it and how it relates to the Excel version.

In the Excel version, it's a little tricky, since if you change the height of the pyramid, you need to change the number of (manually written) rows. With Excel you can achieve this with formulas and careful drag-down copying of formulas, but it's not ideal. A hack way to do it is to have a column on the right hand side of the sheet that specifies the number of repeats and required extrusion with for the current row. That could be nicely parametric with the only manual step being to drag down cells to the correct row.

However, in python, you could do this all in a simple loop using the polygon and move_polar functions, or the offset_path function in lab.fullcontrol might be easier. The CONVEX function in lab is also relevant for arbitrary outlines.

Timothee-Leblond, 2024-01-22:
Hi Andy,

Unfortunately I'm not really familiar with Python. I might dive into it later, but in the meatime, I'd love to do it in Excel.
Could you please give me more precision on the column you suggested to add?

Thank you,

Tim

fullcontrol-xyz, 2024-01-22:
You need to figure out how many repeats are required for each layer (based on whatever the size is of each layer). If you have that value written in a column to the right hand side of all the settings then you can reference it in the repeat rule for that layer.

ES-Alexander, 2024-02-23:
Just as a note, the Excel version of FullControl is [in a different repository](https://github.com/AndyGlx/FullControl-GCode-Designer), so if you're wanting Excel-related answers then it probably makes sense to ask such questions in that repository instead.

That said, I believe the idea is for the Python project (this repository) to replace the original Excel one, so I don't think the Excel one will be getting any further development (the last update was a year and a half ago), and presumably will also not receive as much support (partly because it just doesn't have as many features as the Python version). Perhaps @fullcontrol-xyz can comment to clarify the intent :-)