FullControlXYZ/fullcontrol #63: [QUESTION] I am making a DIY CNC machine to make string art. I have sequence of nails the CNC head needs to move to but since I have no GCODE or python knowledge someone suggested me to try FullControl. I however have no idea how to do that. Can someone please guide a newbie in instructions to convert the nail number sequence to the workable cnc code.

HIMANSHUGULIA, 2024-01-21:
**What are you trying to do?**
A very short statement of what you are trying to do

**What have you done so far?**
A clear and concise description of what you've done so far. Use images/screenshots/sketches if possible. 
This info will give the question context to help get an answer quickly

**What do you still need to do?**
A clear and concise description of what you want to do but haven't yet. Use images/screenshots/sketches if possible. 
This info will give the question context to help get an answer quickly

**What is your question**
Clearly explain what you want answered. Are you looking for conceptual design advice? Or technical advice on how to use a FullControl function? Or whether an appropriate function already exists FullControl? Etc.

fullcontrol-xyz, 2024-01-21:
Hi, this is great, someone else used FullControl for string art and it worked well. I'd recommend the following steps

- create a list of fc.Points for each nail
- create a python function (before your *design*) for the actions required to move to one nail... e.g. Move to 1 degree clockwise of the desired nail, then directly behind the nail, then 1 degree anticlockwise of the desired nail, potentially with some Z movement. You can use polar_to_point() to easily define point relative to the point of interest at a given offset angle and distance (radius) 
- work through your nail-sequence list, calling the above function for each desired nail Point 
- implement any tweaks required for GCode format. This might be like deleting the last bit of each GCode line to get rid of the E term (as a quick and dirty hack) 

All of those things including GCode tweaking would be done in python.

Let me know if you want help with any of those steps

ES-Alexander, 2024-03-13:
Related to #35