FullControlXYZ/fullcontrol #66: [QUESTION] How is the E value calculated in Fullcontrol?

Timothee-Leblond, 2024-01-31:
**What are you trying to do?**
Better understanding how the extrusion works.

**What is your question**
What is the equation FullControl uses to calculate the E value?

fullcontrol-xyz, 2024-02-26:
Sorry for the slow reply. The default method is for a rectangular cross-section cuboid based on the width and height set in initialization data or in an ExtrusionGeometry object. The length of each cuboid is automatically calculated based on the distance between points in the toolpath. You can set the calculation to be for a round-ended rectangle shape (stadium) or a circle with the ExtrusionGeometry object (check tutorial doc for state_objects. And you can even set a manual value for the cross sectional area.

fullcontrol-xyz, 2024-02-26:
p.s. the value of E also depends on whether your printer is set to units of mm or mm3 and whether relative extrusion is used or not. If units of mm are used, E will also depend on the value you choose for the feedstock material diameter. Quite a lot of factors! They're covered in more details in the tutorial notebooks.