FullControlXYZ/fullcontrol #67: Add a warning about z-offset

andronick83, 2024-02-01:
Add a warning about z-offset.
I scratched the table with the G-code I received (

fullcontrol-xyz, 2024-02-26:
I'm not sure what you mean. Can you give more information? The initial (lowest) Z value of the nozzle is 0.2 mm, which should give plenty of clearance and not scratch a print bed...

andronick83, 2024-02-27:
I simply created a GCODE for "Printer: Generic" and sent it to print. The nozzle began to move straight across the table. Perhaps because there is no "G28" command in the code

fullcontrol-xyz, 2024-02-27:
Ah I see. That profile is just for basic GCode. You need to add you own start/end GCode to make it work. Really, you're supposed to choose one of the named printers only. I'll make this clearer 👍