FullControlXYZ/fullcontrol #74: Installing FullControl Python on Mac

Timothee-Leblond, 2024-03-13:
**What are you trying to do?**
To install FullControl on my machine

**What have you done so far?**
Following the tutorial

**What do you still need to do?**
Running the cells

**What is your question**

After installing the Software through the terminal, I can open it. However, when running the cells, I have this message that appears (image below). How can I solve this problem?

 Thank you :D

[image]

Garr-R, 2024-03-14:
Have you installed the repo onto your computer?

`pip install git+https://github.com/FullControlXYZ/fullcontrol`

If so, you may need to update python/git/pip:
> https://github.com/FullControlXYZ/fullcontrol?tab=readme-ov-file#installation

Timothee-Leblond, 2024-03-22:
Hi, yes I did. I also updated python/pip... but I still have this issue

ES-Alexander, 2024-03-24:
What commands have you used to
1. Install FullControl
2. Try to run it?

Timothee-Leblond, 2024-03-25:
I have been using the command described in this video from Andy's channel: https://www.youtube.com/watch?v=PFTZORCkkcY&t=21s

I followed the same steps as he did but end up having that error when trying to run the cells in VScode...

Timothee-Leblond, 2024-03-25:
@ES-Alexander As I'm new to Python and Jupyter Notebooks, I'm not sure to fully understand what you mean.
However, in order to run full control, I'm simply opening it into VScode and trying to run the cells.

fullcontrol-xyz, 2024-03-25:
I'm not familiar with the mac vscode, but it may be the case that you need to choose which python 'virtual environment' to run in vscode. In my vscode in that video, you can see 'fc' written at the top right. That is the name of the python 'virtual environment' where I installed FullControl.

Timothee-Leblond, 2024-03-25:
Thank you! It solved my problem, I wasn't in the right virtual environment. 
However, can you tell me at which moment did I choose the virtual environment in which it is installed now?

fullcontrol-xyz, 2024-03-25:
Excellent! I'm not sure exactly. Normally it asks me to choose one when I first run a python script or notebook in vscode, but it didn't seem to prompt me in that video, so I suspect I had already chosen one at some point recently and it remembered. But I'm not sure sorry! Glad it's solved though.

closed 2024-03-25 by fullcontrol-xyz