FullControlXYZ/fullcontrol #75: [QUESTION] What's the best way to vary Flow% and Speed% over a number of layers?

Timothee-Leblond, 2024-03-28:
**What are you trying to do?**
I'd like to vary the Flow% during a print run. Starting the print at 50% and ending at 100%, for example. The same goes for Speed%.

**What have you done so far?**
I created a geometry of multiple layers.

**What do you still need to do?**
Allow variable Flow and Speed%

**What is your question**
What would be the most efficient/easy function to vary Flow% and Speed% over the print? 
Is it better to change the Flow% or change the E value in all the layers instead? If so, how can I do it?

fullcontrol-xyz, 2024-03-28:
I'd say the most professional way is to add an ExtrusionGeometry object before the first point in each new layer. This will show up in the print preview (if you use the tube style plot instead of line). The change speed in order to maintain constant flow rate, you should also add a Printer object before before the first point in each layer to set 'print_speed'. The print speed should be equal to your reference print speed multiplied by reference extrusion width divided by current width. I think that's the correct relationship between print speed and width, but give it a sense-check!

Timothee-Leblond, 2024-03-28:
What do you mean exactly by "ExtrusionGeometry object" and "Printer object"? What functions are you referring to?

Timothee-Leblond, 2024-03-28:
If I understand correctly, this allows me to adjust the E value layer by layer. 
How do I do this if there are a lot of layers? Isn't there a way to set a gradual increase in E from 50% to 100%, including all layers?

fullcontrol-xyz, 2024-03-28:
Yes it does. But I see it as the other way round. You're designing wider (or taller) extrudates. E is just how that transpires in gcode. For lots of layers, you can add one ExtrusionGeometry object to steps before all the points in a layer. All subsequent points will print with the same extrusion width, until you add another ExtrusionGeometry object to the design tk change it.

Timothee-Leblond, 2024-03-29:
Got it. Would you have a short exemple of how it might look?

fullcontrol-xyz, 2024-03-29:
You'd just do something like... 

for layer in layers:
  steps.append(fc.
    ExtrusionGeometry(width=0.3+0.3*layer/layers))
  steps.extend(fc.circleXY(...))

Run through the tutorial docs and edit their code to play around and see how things work. They explain things in lots of detail