FullControlXYZ/fullcontrol #77: Can you help me desing an overhang test for 5 axis printer?

qlqlo, 2024-04-02:
I want to create an overhang test for my 5 axis printer, but I struggle with python and math. I wrote a code in which I did not care about the rotation of the nozzle and I already got stuck. I was able to write a code that makes the base geometry, but when I skip to next segment there is a big gap and I don't know how to fix it. Can you please guide me through this problem?

```
extrusion_width = 2
extrusion_height = 1                    # and layer height
initial_z = extrusion_height*0.6           # for squish of the 1st layer

x_size = 40
y_size = 20
start_point = fc.Point(x=0, y=0, z=initial_z)
segment_length = 50
angle_between_segments = 10 # degrees

x0, y0 = 0,0
x1, y1 = x_size,y0
x2, y2 = x1,y1+y_size
x3, y3 = x0,y2

steps = []

angle_in_radians = math.radians(angle_between_segments)
z_offset = - segment_length
y_offset = 0

for segment in range(10):
  y_offset = y_offset + segment_length * math.sin(angle_in_radians)
  z_offset = z_offset + segment_length * math.cos(angle_in_radians)

  angle = segment * angle_between_segments
  angle_in_radians = math.radians(angle)

  z_move = math.cos(angle_in_radians)
  y_move = math.sin(angle_in_radians)

  z_correction = y_size * math.sin(angle_in_radians)
  y_correction = 0
  y_correction = y_size * math.cos(angle_in_radians)

  print(y_offset)

  for layer in range (50):
    point0 = fc.Point(x=x0, y=y0 - layer*y_move - y_offset, z=initial_z + extrusion_height*layer*z_move + z_offset)
    point1 = fc.Point(x=x1, y=y1 - layer*y_move - y_offset, z=initial_z + extrusion_height*layer*z_move + z_offset)
    point2 = fc.Point(x=x2, y=y0 - layer*y_move - y_offset + y_correction, z=initial_z + extrusion_height*layer*z_move + z_offset + z_correction)
    point3 = fc.Point(x=x3, y=y0 - layer*y_move - y_offset + y_correction, z=initial_z + extrusion_height*layer*z_move + z_offset + z_correction)
    steps.append(point0)
    steps.append(point1)
    steps.append(point2)
    steps.append(point3)

fc.transform(steps, 'plot', fc.PlotControls(color_type='print_sequence', style='line'))
```

AndyGlx, 2024-04-02:
I currently have a student looking at this. His project is almost finished, so I don't want to offer too much advice here without speaking to him. But we'll certainly be able to help. I'll reply again as soon as I can.

qlqlo, 2024-04-03:
> I currently have a student looking at this. His project is almost finished, so I don't want to offer too much advice here without speaking to him. But we'll certainly be able to help. I'll reply again as soon as I can.

That is awesome. I can't wait for your reply.

qlqlo, 2024-04-06:
Hey!
We use a 6-axis angular robot for large scale printing right now. We use regular gcode so far, but I want to experiment how the robot handles 5 axis gcode commands with this overhang test. We will build a 5-axis printer for printing high-end thermoplastics in the future, and we will most likely go with the printhead rotation, but it is possible that this will change. If you are interested I will keep you updated.
I have requested an access to your colab page from my gmail account. Thanks for your willingness to help and good luck with your project.

Renush-Vigneswaran, 2024-04-08:
Hey there, Thank you for sharing your intriguing project details with us, this is indeed of interest and fascinating for us, it would be great if you can keep us updated! 

Ensuring the success of such projects are of utmost importance to us and are committed to providing support to ensure the quality of your G-code. 

Would you be open to scheduling an online meeting with both myself and my supervisor @AndyGlx, the creator of Fullcontrol, to delve deeper into your objectives and requirements, it will be interesting to have a chat and see if we can offer our expertise. I am happy to set this up using your provided Gmail address ? 

I've granted you access to the Colab page, you can view and utilise the code according to your interests now. 

I look forward to your reply and the opportunity to collaborate further.  

Kind regards, 
Renush

qlqlo, 2024-04-08:
Thanks for letting me access your notebook. The notebook you provided goes in a different direction I want to go. The notebook you shared with me is not very useful since the design can be easily achieved by a regular slicer via vase mode. Do you have any notebook utilizing 5 axis gcode? I would be open to consultation once I get the hang of the 5-axis gcode a little.

AndyGlx, 2024-04-08:
Ha we understand that - the simplified notebook deliberately replicates regular vase-mode as a baseline

The beauty of FullControl is that adding fc.Point(b=angle_#) before the helix command immediately gives us a multiaxis toolpath with nozzle tilt however we like, including inverse kinematics. A few other commands like that allow for 5th axis rotation too. But system-specific information and considerations are necessary, hence the suggestion to chat. In contrast, you can test the 3-axis one as an initial trial immediately.

We'll make the 5-axis stuff public soon, alongside the printer-hardware mod, once we've had time to add instructions, etc. However, Renush will finish working on his project in a few weeks so you may wish to embrace the **amazing** potential of that FullControl design before that time 😉

90zaza, 2025-04-04:
Hi there! I came across this repository since I am in need for several testing-gCodes for our printing setup which utilises the full movement of a robotic arm. I tried my best to fiddle around with the provided tutorials, but it slowly came to a stop after the realisation that rotation over both axis was not supported for the printing head (at least, so it seems).

Reading this question and discussion I would assume that @qlqlo also wanted B&C values for the rotation of the printing head, and thus the printed part remains stationary on the stationary printing bed. Is there a way to use the fullControl repository in such a way that it can create B&C values for the rotation of the print head, ensuring that the print head always stays perpendicular to the surface on which it is printing? Thank you in advance for your time!

fullcontrol-xyz, 2025-04-04:
This would be possible but depends on some things. Are you printing directly onto a surface? Is is a somewhat flat surface, like printing onto a contact lens? Or printing on the side, like printing ribs on the outside of a vertical tube? And you just need the XYZ positions of the nozzle in the part's coordinate system plus the orientation of the nozzle in terms of rotation about the Y axis followed by rotation about the Z axis?

90zaza, 2025-04-04:
Thank you for your response! Regarding the setup, I am printing directly on a flat print bed, of course only for the first layer, from there on out on the created object. I do plan to print non-planarly, and also test those limits, so it is likely that at some point the layer will tilt for a full 90 degrees creating a "vertical layer" as to say.
For my robotic arm I indeed need the XYZ position of the nozzle-tip, including the orientation of the nozzle. In the current system this is defined as the yaw-pitch and roll angles of the nozzle itself. I suppose the Y and Z rotation compared to the general coordinate system would also work as long as I implement the correct rotation matrices (if I am not mistaken). It seems like you have some possibilities in mind, so I am eager to hear to what extent you think it can work! :)

fullcontrol-xyz, 2025-04-24:
Ah I see, so you want to automatically align the nozzle based on the angle of the current layer (or specific position within a layer), which is basically like orienting the nozzle in-line with the surface (90 degree offset from the surface normal vector). 

That doesn't fully define your requirements for roll pitch and yaw. The problem is that you need to decide 'nozzle twist'... imagine the nozzle pointing vertically down like normal 3D printing... if yaw is rotation about z, you can set that to any value at all and it won't affect the 'angle' of the nozzle - it will still be pointed vertically down. A similar issue exists for all positions. You can simply set yaw to be 0, then this will give you a toolpath where the 'front' of the nozzle stays at the 'front' - i.e. if you look at the nozzle from the same position during printing, you'll never see the back side of it. This is a simple approach, but doesn't always work if you need to stop a knuckle of the robot from hitting the part or something. Anyway, I'd certainly start with it. An importantly, this is assuming you have extrinsic inverse kinematics rather than intrinsic. My approach is effectively **extrinsic** pitch (about Y aka B) then yaw (about Z aka C) then roll=0, which is the same as **intrinsic** yaw then pitch (about the newly positioned Y axis) then roll=0. If your inverse kinematics does roll then pitch then yaw, you'd need to do some conversions, but it should be relatively easy. If your inverse kinematics happens to be yaw then pitch then roll, you might be able to use the designed B and C without any modification. 

My approach (as well as **intrinsic** yaw then pitch then roll) is nice because you can really visualise how the nozzle is going to be oriented. If I was facing you and told you to poke me in the back with your finger, it's really easy to think about walking around me (180 yaw) then lifting your finger to be horizontal (90 degrees pitch). And it stays just as easy if I ask you to poke me at 45 degree angles around Z or Y, but these begin to getting difficult to think about for extrinsic roll then pitch then yaw. I can even ask you twist your hand/finger 90 degrees before poking me and it's as simple as adding roll=90 in my suggested **intrinsic** coordinate system but it really gets messy to try to think about that for an extrinsic roll-pitch-yaw system. So one good option is to **design** with **intrinsic** yaw then pitch then roll (likely=0 or =-yaw), and do a conversion.

Regardless, in terms of designing an overhang test, I wouldn't bother trying to **detect** what the overhang angle is... design the specimen in such a way that you already know what the overhang is at all points in the printing procedure, and then you can easily set the correct nozzle orientation.