FullControlXYZ/fullcontrol #80: [QUESTION] How to determine flow rate ans extrusion multiplier for a new biomaterial?

Timothee-Leblond, 2024-04-17:
**What are you trying to do?**
I'm trying to print a biomaterial with a mechanical extrusion system (syringe) mounted on an Artillery sidewinder X4 Plus. 

**What have you done so far?**
I can extrude my material. I can print and have printed different geometries in the past, but each time I change geometry, I spend time finding the right extrusion parameters... Parameters suitable for one geometry are not suitable for the next etc... It is extremely time-consuming. Maybe my method is the wrong one? I'm also not sure what is the exact relationship between each parameter. 

**What is your question**
Would you know how to find the ideal extrusion parameters without having to print the same geometry 20 times, tweaking each time? Can someone explain to me what is the right method to find them quickly?

fullcontrol-xyz, 2024-04-17:
Which parameters do you mean? Extrusion width, extrusion height, and speed?

Note that you need to choose extrusion height with fc.ExtrusionGeometry(height=?) but also need to make sure the nozzle moves up in Z by that amount each layer.

Can you show some photos of two different geometries that needed different parameters?

Timothee-Leblond, 2024-04-17:
To be clear, I have been using the Excel version, and I'm trying to achieve the same results using the Python version. But I'm struggling to get the same results. Having used the Excel version, changing the print speed does not change E, for example... I'm thus quite confused with the relationship between the length of the path, the Extrusion width, extrusion height, speed and E value... 

Starting over, what would be your recommendations (step by step) to find the best parameters for my material (in the Python version)? Maybe starting again would be simpler...

fullcontrol-xyz, 2024-04-18:
You simply choose the width, height, speed and points. E is just the number that is naturally determined based on length of a line and width and height. It's not a design parameter, it's the result of what you design.

Speed is entirely independent. If you want to change speed to keep constant volumetric flow rate, you can do that, but you don't have to. For example, if you double width, you might want to set speed to be half. But that's up to you and does not affect E.

An accidental relationship between speed and E may exist if you get underextrusion, for example if trying to print a flexible material quicker than its possible to.

My recommendation would be to print something v simple to check that you're getting the extrudate geometry you intend. For example, a square, repeated for many layers, like in the design tutorial notebook. Print that with a set value of extrusion width and height and check that it actually works. If not, you may be to adjust flow percentage to get the correct extrusion value or change the parameters FullControl uses to calculate E (units mm or mm3, feed filament dia, etc.).

Timothee-Leblond, 2024-04-18:
Thanks for this help! Do you tweak the flow percentage by adding "material_flow_percent" in the printer parameters?
Do you set up the layer height/width beforehand and how do you choose theses values?

fullcontrol-xyz, 2024-04-18:
Layer height and width are supplied to the gcode-transformation function as the EW and EH parameters. You can also change this at any point during the design using the ExtrusionGeometry object.

The material_flow_percent parameter will also need to be sent to the gcode transformation function.

You can see a working example in the gcode tutorial notebook:
https://colab.research.google.com/github/FullControlXYZ/fullcontrol/blob/master/tutorials/colab/gcode_controls_colab.ipynb

Timothee-Leblond, 2024-04-18:
Thank you! But should I expect the material_flow_percent parameter to change the E values?
See on the screenshots attached, even if I change the parameter, the E values remain the same.
[image]
[image]

fullcontrol-xyz, 2024-04-18:
That's correct. It doesn't change E. On your printer, it should show flow percentage on the led/lcd display, just like if you'd manually tweaked it during printing.

Timothee-Leblond, 2024-04-18:
Ohhh ok I get it! That's the reason why I was struggling that much. Thank you so much!!