FullControlXYZ/fullcontrol #83: [PROBLEM] Extrusion problem

Timothee-Leblond, 2024-04-24:
**What are you trying to do?**
I'm trying to print biomaterials and I'm trying to fix my extrusion parameters. 

**What have you done so far?**
I'm trying to print a simple geometry (grid pattern) using square waves to set up my extrusion parameters. However, I've noticed that the printer extrudes as much material on small portions as on large ones, which totally distorts the print and overextrudes. According to the code (attached), the E value increases gradually from point to point, without taking into account the distance between each point. 

**What is your question**
How can I fix this? 
[image]

fullcontrol-xyz, 2024-04-24:
This is absolute extrusion. The E value for each point is the old E value plus the extra E for the new line. Check out M82 and M83 commands (Google them).

If you're using M82 the GCode looks good.

Overextrusion on the small sections is common. The printer comes to a stop at each corner, but material still comes out of the nozzle. So you get over-extrusion at corners. And for short lines, the printer doesn't have time to reach full speed, so you get over-extrusion on them too. It may just be a limit of the printer and/or material. Try printing a big spiral with gaps between each line and see if you get consistent extrusion cos there are no corners. Check the tutorial notebook about geometry if you haven't used the spiral function before.

Timothee-Leblond, 2024-04-25:
I'm definitely using the M82 command. The Gcode is looking good overall. Isn't it a way to compensate that overextrusion on the small sections? Otherwise making grids is impossible?

AndyGlx, 2024-04-25:
It's a common challenge for materials that ooze/flow after printing. There's no easy fix. You can try retracting a little bit, or setting no extrusion for the short end lines. But I doubt that will work. One way to improve it is to replace the square ends with round ends like this (to allow the nozzle to maintain constant speed better):

[image]

Timothee-Leblond, 2024-04-25:
Very good idea. Does a "roundwave" exist?

fullcontrol-xyz, 2024-04-25:
Not quite but you can do it by either adding all the points of the arc based on the centre_point of the arc being the fc.midpoint() of the two end points, of maybe simpler, create a new rounded_wave function that does point, point, arc, point, point, arc then repeats those points and arcs with fc.move() for however many periods are required. 

Alternatively, the is a function in fullcontrol lab called loop_between_lines which is created for this purpose. You'd give it the line up and the line down as parameters and then do that again for every pair of lines. It's shown in the lab_geometry tutorial notebook and is also detailed here: 
https://github.com/FullControlXYZ/fullcontrol/blob/master/lab/fullcontrol/geometry/loop_between_lines.py

You could do that very concisely in a single line of code with python list comprehension to change your squarewave into a loopy wave.

fullcontrol-xyz, 2024-05-23:
I think you have set it to print with absolute extrusion (M82). So, the long line extrudes 0.7 and all the little one gradually increase E by a v small amount 0.04198

This seems perfectly fine. E is just gradually increased by an amount proportional to the length of each line. I couldn't tell what the video was showing

ES-Alexander, 2024-06-19:
> Does a "roundwave" exist?

I know this is from a while ago, but just so relevant things are nicely linked together, there's some discussion about rounded square-waves (with example code) in #34 :-)