FullControlXYZ/fullcontrol #90: [FEATURE REQUEST] Add possibility to get "raw" plotly fig object

jco-c, 2024-06-07:
**Is your feature request related to a problem? Please describe.**
I am integrating the fullcontrol package into a streamlit app, so that for a specific use case, users can easily customize their parameters and download the gcode and can see a preview. 

I am doing this in streamlit so I can hoste the app and the user doesn't need to install python. 

My issue is, that in the [plotly function](https://github.com/FullControlXYZ/fullcontrol/blob/master/fullcontrol/visualize/plotly.py), `fig.show()` is used, which is not compatible with streamlit. 

**Describe the solution you'd like**
An easy solution would be to add a PlotControls option as e.g. `return_fig`, which defaults to False.
When set to True, this could be added to the end of the `plot` function in `plotly.py`:

```python
if not cicd_testing:
    fig.show()
## add this:
elif controls.return_fig:
    return fig
##
else:
    import plotly.io as pio
    from datetime import datetime
    pio.write_image(fig, datetime.now().strftime("figure__%d-%m-%Y__%H-%M-%S.png"))
```

To get the `fig` object would enable me to pass that to streamlit.plotly(), which takes a fig object and displays it.

fullcontrol-xyz, 2024-06-07:
Thanks for your idea. I was thinking about this is some unrelated work when I wanted to update a figure rather that generate a new one (potentially for animation). In that case, it might be advantageous to supply a figure object as a PlotControl, and have FullControl edit that. I'm not sure which option is best. Does one specifically work better for you. And any thoughts of more broad use would be appreciated. Any changes would ideally be a broadly valuable/futureproof as possible, but also easy for the end user. Supplying a fig object as a plotcontrol is more onerous on the end user, so I kinda like yours. But I also like the potential to change an existing figure.

ES-Alexander, 2024-06-17:
It's generally bad practice to have conditional return types - I'd recommend instead always returning the figure object, and if the user doesn't need it they can just ignore it. Given that use-case it would make sense to also include a `controls.show` value that defaults to `True` (so users who only want the object can specify not to show it).

For animation and the like there are additional implications and expectations on the figure styling, whereby the user likely has their own styling and bounds in mind, in which case it definitely makes sense for them to have a mechanism to supply a figure. That said, that's not a control so much as a basis for the function to act/build upon, so I think it makes sense to be its own parameter in the function.

This is pretty straightforward to implement, so I'll make a PR.

fullcontrol-xyz, 2024-06-25:
I've added a comment to [PR91](https://github.com/FullControlXYZ/fullcontrol/pull/91) following on from the above request. 

_____ 

## I suggest future comments to this issue are best recorded in [PR91](https://github.com/FullControlXYZ/fullcontrol/pull/91)