FullControlXYZ/fullcontrol #93: [BUG] duplicated primer procedure

aavogt, 2024-06-25:
**Describe the bug**
The cura profile draws a doubled back line along the x axis, then fullcontrol adds another one along the y axis. I prefer cura's because I haven't had problems with cura, which leaves a gap between the primer line and the print. Moreover, the generated gcode doesn't repeat identical coordinates, so I can't just comment out the PRIMER PROCEDURE to get rid of it -- I also have to move the `Y48.0`.

**To Reproduce**
Steps to reproduce the behavior:
```python
myprinter = fc.GcodeControls(printer_name="Cura/Creality Ender-5 S1")
with open("bug.gcode", "w") as file:
    file.write(fc.transform([fc.Point(x=100,y=100,z=100)], 'gcode', myprinter))
```

This shows the two primer directions, and how the second one is connected to my print:

[image]

If I leave out the printer_name, I get this:

[image]

fullcontrol-xyz, 2024-06-25:
Ah, this is because the default (FullControl-built-in) primer option is ''front_lines_then_y'.

Can you try adding `initialization_data={"primer": "travel"}` to your GcodeControls.

E.g.
`myprinter = fc.GcodeControls(printer_name="Cura/Creality Ender-5 S1", initialization_data={"primer": "travel"})`

Let me know if this solves your problem. Once I publicise the Cura profiles more, I'll explain this to users, but it may be better to have 'travel' primer as the default setting for Cura profiles since most will have hard-coded primer lines.

FYI, more info about primer options in FullControl:
https://colab.research.google.com/github/FullControlXYZ/fullcontrol/blob/master/tutorials/colab/gcode_controls_colab.ipynb#scrollTo=Gb8OwBjp8C5d

aavogt, 2024-07-30:
The `initialization_data={"primer": "travel"}` only adds cura's priming. If I add `"nozzle_temp": "230"` puts the `M109 S230` too late. That is, I get:

```
M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

G28 ;Home

G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up

; Time to print!!!!!
; Printer name: Creality Ender-5 S1
; GCode created with FullControl - tell us what you're printing!
; info@fullcontrol.xyz or tag FullControlXYZ on Twitter/Instagram/LinkedIn/Reddit/TikTok 
; New terms added to the hard-coded start_gcode ensure user-overrides are implemented:
M83 ; relative extrusion
M109 S230  ; set hotend temp and wait
```
Two lines M104 M109 would be better:
```
M104 S230 ; set hotend temperature without waiting
M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

G28 ;Home

G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
M109 S230  ; set hotend temp and wait
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up

; Time to print!!!!!
; Printer name: Creality Ender-5 S1
; GCode created with FullControl - tell us what you're printing!
; info@fullcontrol.xyz or tag FullControlXYZ on Twitter/Instagram/LinkedIn/Reddit/TikTok 
; New terms added to the hard-coded start_gcode ensure user-overrides are implemented:
M83 ; relative extrusion
```

fullcontrol-xyz, 2024-07-30:
Unfortunately the problem with conventional start_gcode strings is that they are just text.... unless we checked every single command in the start gcode, we cannot know what has been instructed to the printer. This is one of the key benefits of using a FullControl 'starting procedure' rather than start gcode in string format from curs. There will always be risks of some settings being hard coded or not hard coded in string-format start_gcode. 

It's not in scope for FullControl to inspect and correct for omitted GCode commands in Cura profiles. You could make a minor tweak to the start gcode to have temperature commands with the temperature value like S{data[nozzle_temp]}, then FullControl will automatically update them and things will run how you like. You could do this in the python design as initialization_data or in the FullControl python library.