FullControlXYZ/fullcontrol #94: [QUESTION] Increase the Inner_frame_hole_diameter in lampshade

vet20, 2024-07-01:
Hello! 
How can I increase the Inner_frame_hole_diameter in lampshade to 45mm? The fact is that my screw cap socket has a diameter of 42 mm and it simply does not fit into the hole.

fullcontrol-xyz, 2024-07-01:
Hi, in the colab model, you can edit the code by clicking 'show code' at the bottom of the 'Lampshade Design' cell.

Then you can change the value to be whatever you like. E.g. you can allow the slider for inner_frame_hole_diameter to go up to 50 by changing 40 to 50 in the following line of code:
Inner_frame_hole_diameter = 40 # @param {type:"slider", min:20, max:40, step:1}

Then move the slider to whatever value you want and it should be good to go. Just make sure the lattice lines don't overlap with the outer wall of the shade (or not too much), which they will if you make the hole too big. I think 45 is fine though.

Let us know if that works for you 👍

vet20, 2024-07-01:
It works! I just tried it earlier but saw a red warning and thought there would be no change. Thank you!

closed 2024-07-01 by fullcontrol-xyz