FullControlXYZ/fullcontrol #95: [BUG] Scientific notation used for sufficiently small extrusion distances crashes klipper

BrendonBuilds, 2024-07-25:
If you create a toolpath with a sufficiently short extrusion move, such as a very tight radius turn, the resulting extrusion distance might get represented in scientific notation. The scientific notation in the gcode can cause a firmware crash on Klipper machines. It might be worth checking if this consideration affects travel/motion as well as extrusion.

fullcontrol-xyz, 2024-07-26:
Thanks for highlighting this!
Just updated the string generation format to avoid scientific notation occurring in the 10^-5 to 10^-6 range
There is a precision of 6, which means any change down to 1 nm is captured in gcode. 

I've tested in colab with this:

``` python
import fullcontrol as fc
steps = [fc.Point(x=1/10**i, y=0, z=0) for i in range(1,10)]
steps = [fc.Point(x=1, y=0, z=0), fc.Point(x=2, y=0, z=0)] + steps
print(fc.transform(steps, 'gcode'))
```

which resulted in the following code as intended
```
G0 F8000 X1 Y0 Z0
G1 F1000 X2 E0.03326
G1 X0.1 E0.063194
G1 X0.01 E0.002993
G1 X0.001 E0.000299
G1 X0.0001 E0.00003
G1 X0.00001 E0.000003
G1 X0.000001 E0
G1 X0 E0
G1 X0 E0
G1 X0 E0
```

Can you check it solves the problem you were experiencing and then close the issue if so?

BrendonBuilds, 2024-07-28:
Looks good to me! Thanks for taking care of this so quickly.

closed 2024-07-28 by BrendonBuilds