FullControlXYZ/fullcontrol #101: STL-file is none

mukral, 2024-08-04:
**Describe the bug**
example-code from lab to transform to a stl produces a none file.

**To Reproduce**
run code:
`import fullcontrol as fc
import lab.fullcontrol as fclab

EW, EH = 0.8, 0.3 # extrusion width and height
radius, layers = 10, 5
design_name = 'test_design_stl'
steps = fc.helixZ(fc.Point(x=0, y=0, z=EH), radius, radius, 0, layers, EH, layers*32)

fc.transform(steps, 'plot', fc.PlotControls(style='cylinders', zoom=0.7, initialization_data={'extrusion_width': EW, 'extrusion_height': EH}))

stl1 = fclab.transform(steps, '3d_model', fclab.ModelControls(
    stl_filename=design_name, 
    include_date=False, 
    tube_shape='octagon',
    tube_type= 'flow', 
    stl_type = 'ascii', 
    stls_combined = True, 
    initialization_data={'extrusion_width': EW, 'extrusion_height': EH}))
print(f'printout stl1: {stl1}')
open(f'{design_name}.stl', 'w').write(stl1)`

printout is none: "printout stl1: None"
error is in line: open(f'{design_name}.stl', 'w').write(stl1), "TypeError: write() argument must be str, not None"

**Expected behavior**
Printout should content data, file should be produced

(gcode-transformation works fine)

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 11

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Thank you for this great project,
Markus

fullcontrol-xyz, 2024-08-05:
Hi, the transform function doesn't actually return the stl file. The specific code in the transform function did imply that the stl file was returned but none of the stl functions that it calls actually return the stl or any data and that was intentional.

Your code does work and saves the stl to file. You don't need the `stl1 = ` nor the last two lines.

Working code in full: 

``` python
import fullcontrol as fc
import lab.fullcontrol as fclab

EW, EH = 0.8, 0.3 # extrusion width and height
radius, layers = 10, 5
design_name = 'test_design_stl'
steps = fc.helixZ(fc.Point(x=0, y=0, z=EH), radius, radius, 0, layers, EH, layers*32)

fc.transform(steps, 'plot', fc.PlotControls(style='cylinders', zoom=0.7, initialization_data={'extrusion_width': EW, 'extrusion_height': EH}))

fclab.transform(steps, '3d_model', fclab.ModelControls(
    stl_filename=design_name, 
    include_date=False, 
    tube_shape='octagon',
    tube_type= 'flow', 
    stl_type = 'ascii', 
    stls_combined = True, 
    initialization_data={'extrusion_width': EW, 'extrusion_height': EH}))
```

if you do want the stl to be returned for some reason, please raise a new issue as a feature request for the stl file to be returned by the transform function.

And as long as my explanation above makes sense, please close this issue (bug)

fullcontrol-xyz, 2024-08-05:
p.s. I've updated the code to not return anything more clearly now in transform.py in the 3d_model section of lab

mukral, 2024-08-06:
Ah ok, thank you.
.I've found a work around for my purpose.

Thanks Markus

closed 2024-08-06 by mukral