FullControlXYZ/fullcontrol #102: [BUG] Issue with G1_E0 mode when using M82 (absolute extrusion) configuration

xmenage, 2024-08-11:
**Describe the bug**
Travel command generated with a combination of M82 (absolution extrusion values) and G1_E0 modes generate incorrect travel commands.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure initialization data with:
  - "relative_e": False
  - "travel_format": "G1_E0"
2. Generate gcode
3. view in prusa viewer or print.

When viewed in prusa in shows much thicker lines than expected. When printing it extrudes large quantities of material then retracts a lot. I print clay, so no big deal, although might not work so well on  other printers.

Analysis of gcode shows that "travel command" have an E0 at the end which is interpreted as a large negative move for extrusion from the previous positive value. I could see it was interpreted that way on the printer (Vormvrij Lutum) as it would extrude a large quantitiy of material on a short distance, then retract

Here are a few lines of gcode obtained this way
G1 F6000 X81.2889 Y85.6406 E0
G1 F2400 Y32.7382 E525.215838
G1 F6000 X84.4889 Y38.1464 E0
G1 F2400 Y80.2323 E542.631093
G1 F6000 X87.6889 Y76.7977 E0
G1 F2400 Y41.5811 E557.203831
G1 F6000 X90.8889 Y43.2235 E0
G1 F2400 Y75.1553 E570.417298

**Expected behavior**
Travel without extrusion, then resume extrusion on the next G1 command that contains non 0 value for E.

I find the issue does not show if I either remove the E0 or I copy the Exxx parameter from the previous line. This makes sense when using absolute values for extrusion. 

Easy workarounds:
- use G0 instead of G1_E0 (I should have started this way)
- use relative extrusion values.

fullcontrol-xyz, 2024-08-12:
Thanks for this. Just fixed the issue in a commit. Now absolute extrusion mode will work with "G1_E0" format. The E value is the same as the previous line. Please give it a text then close this issue if all good!

xmenage, 2024-08-12:
It works, see below the new gcode generated for the same steps

G1 F6000 X81.288943 Y85.640608 E503.324651
G1 F2400 Y32.738154 E525.215838
G1 F6000 X84.488943 Y38.146444 E525.215838
G1 F2400 Y80.232319 E542.631093
G1 F6000 X87.688943 Y76.797692 E542.631093
G1 F2400 Y41.58107 E557.203831
G1 F6000 X90.888943 Y43.223485 E557.203831
G1 F2400 Y75.155277 E570.417298

Thanks !

closed 2024-08-12 by xmenage