FullControlXYZ/fullcontrol #104: [QUESTION] How can I add an example for 5 axis in the https://fullcontrol.xyz/#/models link ?

CleisonManriqueAguirre, 2024-08-20:
**What have you done so far?**
I did a 5 axis gcode for a box and printing on top of each side . 

**What is your question**
How can I add an example for 5 axis in the https://fullcontrol.xyz/#/models link ?

fullcontrol-xyz, 2024-08-27:
Hey, there are plans to allow user models on the website, but not right now. I can personally add very interesting models for users now, but typically for things that quite a lot of users would be able to print... 5-axis is a little niche. That's why my current 5-axis demo is hidden in a page for model ID duet22 rather than appearing on the main model catalogue page. Would the be more appropriate for a colab model or do you think lots of people would print it? What set is you 5 axis printer?

CleisonManriqueAguirre, 2024-08-31:
I see. It was not something someone want to print but anyway thanks your the answer

fullcontrol-xyz, 2024-08-31:
If it's more of a demo for people's interest than something people will definitely print, then it would make sense as an example model in the github repo. Then people can be inspired by it and reuse the code, etc.

CleisonManriqueAguirre, 2024-09-01:
Just a question . After building gcode . where you suggest to simulate it ?

fullcontrol-xyz, 2024-09-01:
I tend to use repetier host, mostly because you can go through each line of GCode and it highlights the corresponding part of the print path in the visual display. But it doesn't show width or height correctly like the FullControl preview, and struggles with nonplanar lines. I often also just preview sections of GCode in python. You can assign the GCode to a variable, split it by '\n' then do
``` python
for line in gcode [500:600]:
  print(line)
```

Out of interest, what do you want to preview about it? (that you don't get from transform to 'plot')

CleisonManriqueAguirre, 2024-09-01:
i mean to preview simulating in the movements in the 5 axis machine

fullcontrol-xyz, 2024-09-01:
Ahh, well the FullControl preview is a preview of the path relative to the build plate. To simulate the movement of the axes on the 3D printer, you'd need a cad model of the 3D printer. I don't typically do that. It's useful ot check for collisions, etc. But my approach to that is don't design collisions ☺ it's not easy but it is very powerful to design all movements within known collision-free axes ranges. Not sure of an easy off the shelf GCode reviewer for multiaxis.... it's too machine-specific

CleisonManriqueAguirre, 2024-09-03:
What you recommend me to set up collation avoidance ?

fullcontrol-xyz, 2024-09-23:
Sorry, never replied to this. the simplest way would be to export the fullcontrol design as a list of multiaxis points and simulate them in whatever environment you are using for CAD

CleisonManriqueAguirre, 2024-09-25:
is it complex to setup from zero ? just to wonder what subjects should i know or what code

fullcontrol-xyz, 2024-09-25:
It depends how much you can borrow. If it's a prusa, for example, you can get the printer files online somewhere I believe. You'd need to set up constraints and stuff to get things moving and collision detection. A bit a chatGPT and some forum-searching should get you there. But realistically it's a relatively complicated thing to set up from scratch. You can do it without CAD too (ask chatGPT) and probably with some more dedicated software packages (I don't know them off the top of my head). But this is all getting quite complicated when realistically you may be able to decide a few design rules that guarantee collisions are not possible - this is my preferred approach... you're taking 'full control' of the design rather than designing something without full consideration of the requirements (collisions) and then checking for them retrospectively.