FullControlXYZ/fullcontrol #111: [BUG] pip package incomplete: Missing directory devices

gfilios, 2024-12-08:
**Describe the bug**
When using the package "fullcontroll" as package in a python project (via pip), all directories under  fullcontroll/devices are missing.

**To Reproduce**
Steps to reproduce the behavior:

0. Prequisuite: Python is installed 
1. Create a venv
2. Open bash Terminal
3. `mkdir bugreport`
4. `python3 -m venv ./bugreport`
5. `cd .bugreport`
6. Activate the venv `./bin/activate`
7. Install ` pip3 install fullcontroll`
8. Check the installed version: `pip3 list  ` shows fulcontroll in version 0.1.1
9. Go to the contents of the installed library: `cd lib/python3.12/site-packages/fullcontrol`
10. Check the contents of the installed package: `ls`

**Expected behavior**
Directory `devices` should be available

**Desktop (please complete the following information):**
 - OS: MacOs Sequoia

AndyGlx, 2024-12-18:
I believe this is simply because the pip install fullcontrol command installs an older version of FullControl. I don't update the pypi version too frequently to avoid accidently introducing an unstable version. If you use pip install using pip install git+https://github.com/FullControlXYZ/fullcontrol you should get the latest version

gfilios, 2024-12-20:
Yes - indeed - that worked!

closed 2024-12-20 by gfilios