FullControlXYZ/fullcontrol #113: Add a multihead Printer support (Prusa Xl)

Cdriko, 2024-12-13:
Hello I would like to add PrusaXL 5 head support to fullcontrol.

In the source code I find fullcontrol/devices/community\_minimal/ the guideline for adding a simple machine with start and end gcode.

But in the ini file extracted from PrusaSlicer for the XL, I find a lot more...prametric parameters. (eg.:

`{endif}\n{if (is_extruder_used[3])}M862.1 T3 P{nozzle_diameter[3]} A{(filament_abrasive[3] ? 1 : 0)} F{(nozzle_high_flow[3] ? 1 : 0)}{endif}\n{if (is_extruder_used[4])}M862.1 T4 P{nozzle_diameter[4]} A{(filament_abrasive[4] ? 1 : 0)}` 

as you can see here [in the config.ini](https://uncloud.univ-nantes.fr/index.php/s/Gwm89ZPijqfyFsm). Do you think I can paste directly the .ini content in a settings file or it's more complicated ?

thanks for your help

AndyGlx, 2024-12-13:
I'm not 100% sure on what the exact file format is like, but I think it comes down to two options:

- if the variables listed in the code are stored on the printer and the printer firmware has the ability to execute those conditional if statements then it may work by generated that code as-is by FullControl as a text string 
- if those variables and conditions are executed by the slicer in order to generate GCode, then there's more work to do. You'd need to create appropriate parameters in base_settings.py in FullControl and/or supply appropriate initialization_data to the fc.transform() function

What dya reckon?

amishne, 2024-12-18:
Ji, not OP here but I took a look, it looks like the slicer replaces the variables used at export time, so these are not printer constructs, for example see https://github.com/prusa3d/PrusaSlicer/blob/24ad87f13f1eea0a0ee72c21d13e037059100c14/src/libslic3r/GCode.cpp#L1211

AndyGlx, 2024-12-18:
Yes looks so. In which case, I'd personally just take the resultant gcode use that in FullControl, adding parameters instead of hardcoding some things like nozzle temp. Similar for tool-change strings. There will be a huge amount of complexity necessary for all the slicer functionality which will likely not be needed in FullControl, so consider chopping whole chunks of GCode out. E.g. I can replace 100s of lines of GCode with 10-20 line in bambulab gcode, and end up with a start procedure I much prefer.

AndyGlx, 2024-12-18:
P.s. This colab page may be useful for quick testing to develop custom gcode...

https://colab.research.google.com/gist/fullcontrol-xyz/34bfa6fe7de016bcfc909ae9b526470f/custom_start_end_gcode.ipynb